deneyapkart/deneyapkart-arduino-core #5: DeneyapKart Kamera'yı ESP32 Cam koduna uyarlama.

KaptanINO, 2021-08-19:
Merhaba. Ne yazık ki günümüzdeki ipv4 adres kıtlığı sebebi ile port açmak mümkün değil. Ngrok, portmap.io ve packetriot gibi servisler ise verimli çalışmıyor hatta hiç çalışmıyor. Şu an bunun tek çözümü websocket ile dış bir server ile haberleşme kurarak kamera'yı dış dünya bağlantısını açmak. İnternette ESP32 Cam ile çalışan (bkz. https://www.youtube.com/watch?v=kE0idrJYPII)
bir proje var. GCS kullanarak sunucu açıyoruz ve websocket ile bağlantı kuruyoruz. Orda verdikleri kod ESP32 Cam için geçerli.
Peki o projede kullanılan kod (bkz. workspace: https://github.com/0015/ThatProject/tree/master/ESP32CAM_Projects/ESP32_CAM_LocalServer/CameraWeb kod: https://github.com/0015/ThatProject/blob/master/ESP32CAM_Projects/ESP32_CAM_LocalServer/CameraWeb/CameraWeb.ino)DeneyapKart ve kamerası ile çalışır mı? Ne gibi düzenlemeler yapmamız lazım? Yardımcı olmanızı rica ediyorum. İyi günler

ozgurbostan, 2021-08-22:
Merhabalar,

Belirtmiş olduğunuz projeyi Deneyap Kart ve Deneyap Kamera ile kullanabilirsiniz. İlgili Web Socket kütüphanesini yükledikten sonra ilişkili olan kod satırlarını ekleyip kolaylıkla kodunuzu oluşturabilirsiniz gibi görünüyor.

void loop() fonksiyonu altındaki gerekli değişiklikleri de yapmayı ihmal etmeyiniz.

KaptanINO, 2021-08-22:
> Merhabalar,
>
> Belirtmiş olduğunuz projeyi Deneyap Kart ve Deneyap Kamera ile kullanabilirsiniz. İlgili Web Socket kütüphanesini yükledikten sonra ilişkili olan kod satırlarını ekleyip kolaylıkla [...]

Yani deneyap kütüphanesini eklememe gerek yok. Malum kodun içinde kameraya ait pinler belirtilmiş.

ozgurbostan, 2021-08-23:
Deneyap kütüphanesi sistemde varsayılan olarak arka planda ekli zaten, sizin ayrıca eklemenize gerek yok.

closed 2021-08-23 by KaptanINO