deneyapkart/deneyapkart-arduino-core #10: A4,A5,A6 Pins Seems Not Running As Analog Input

acarmetitu, 2021-11-21:
At version 1.3.0,  A4, A5 and A6 pins are not working as analog input. A0,A1,A2 and A3 seems working as expected but, analogRead() always return zero for A4,A5 and A6 pins.

zeynepdicle, 2021-11-22:
There is no A6 pin.  Can you share the your code for A4 and A5?

acarmetitu, 2021-11-22:
My comments are related to Deneyap Mini Board.  Sample code is given below.

```
uint16_t a, b, c, d, e, f, g;

void setup()
{
  Serial.begin(115200);

  pinMode(A0, INPUT);
  pinMode(A1, INPUT);
  pinMode(A2, INPUT);
  pinMode(A3, INPUT);
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);
  pinMode(A6, INPUT);

}

void loop()
{
  a = analogRead(A0);
  b = analogRead(A1);
  c = analogRead(A2);
  d = analogRead(A3);
  e = analogRead(A4);
  f = analogRead(A5);
  g = analogRead(A6);

  Serial.print(a); Serial.print("\t");
  Serial.print(b); Serial.print("\t");
  Serial.print(c); Serial.print("\t");
  Serial.print(d); Serial.print("\t");
  Serial.print(e); Serial.print("\t");
  Serial.print(f); Serial.print("\t");
  Serial.println(g);

  delay(50);
}
```

zeynepdicle, 2021-11-23:
Error fixed and new v1.3.1 Release are out. You can try.

closed 2021-11-23 by acarmetitu