deneyapkart/deneyapkart-arduino-core #11: Wifi modülünde sorun

gokturkDev, 2021-12-15:
Merhabalar, 

Deneyap kartımda bir web sunucusu oluşturmak istedim.
Sitedeki https://deneyapkart.org/proje-118.html bu projedeki kodu, Android IDE me yapıştırdım ve compilelamayı denedim ancak şu hata ile karşılaştım

[image]

Biraz incelemeden sonra WifiESP32.h nin gerçekten bu dizinde olmadığını farkettim
[image]

Daha sonra internette farklı bir çözüm arayışına girdim ve şu tutorial i buldum
https://randomnerdtutorials.com/esp32-web-server-arduino-ide/
Farkettim ki bu kod, deneyap sitesindeki tutoriali oluşturan kodun boilerplate'i. 
Bu kodu Android IDE me yapıştırdım ve compilelamayı denedim ve şu hatayı aldım.

[image]

Biraz inceleme sonucu, IDE deki boardu, deneyap karttan ardunio UNO ya değiştirdim ve bu şekilde compilelamayı denedim ve başarılı bir şekilde compileadım

[image]

Anladığım kadarıyla, deneyap reposu Wifi modülünün outdated, eksik veya hatalı bir versiyonunu kullanmakta. Lütfen bu konuyla ilgilenebilir misiniz?

Çok teşekkürler,
Kolay gelsin

zeynepdicle, 2021-12-21:
Merhabalar, 
Arduino IDE'de Deneyap Kütüphanesinin kaçıncı versiyon yüklü ?

gokturkDev, 2021-12-22:
Merhaba,

Bu issueyu açmadan önce, deneyap kartın sitesindeki tutoriali çalıştırmayı denemek için daha eski bir deneyap kart versiyonuna (1.1.0) geçmiştim. Bu issueyu yazarken https://randomnerdtutorials.com/esp32-web-server-arduino-ide/ bu kodu bu versiyonda çalıştırmışım

Updateleyip versiyon 1.3.2 e geçince https://randomnerdtutorials.com/esp32-web-server-arduino-ide/ buradaki kod sorunsuz bir şekilde çalıştı
Ancak deneyap kartın sitesindeki tutorial yine derleme hatası verdi.
https://deneyapkart.org/proje-118.html

İlginiz için teşekkür ederim,
Kolay gelsin

closed 2021-12-22 by gokturkDev