deneyapkart/deneyapkart-arduino-core #15: Deneyap Mini Kart ile RC522 Modülünün Kullanımında Problem

MrDeadLine, 2022-10-29:
### Hardware:
Board | : | Deneyap Mini Kart
----- | - | ------------
Core version | : | 1.3.8
IDE | : | Arduino IDE 2.0.1
Computer OS | : | MacOS 12.6

### Description:
Deneyap Mini kartımla RC522 RFID modülünü kullanarak 13.56MHz RFID kart(lar)ımı okuyup seri monitörde yazdırmaya çalışıyorum. Ancak RFID kartı modüle yaklaştırdığımda hiç yaklaştırmamışım gibi seri monitörde okunmadı bilgisi görüntüleniyor.
Yaptığım pin bağlantısı:
RC522 -> Deneyap Mini
SDA -> D2
SCK -> SC (D7)
MOSI -> MO (D4)
MISO -> MI (D5)
GND -> GND
RST -> D3
3.3V -> 3.3V

### Sketch to reproduce the error:  
```cpp
#include <SPI.h>
#include <MFRC522.h>

#define RED LEDR
#define BLUE LEDB
#define GREEN LEDG
#define RST_PIN D3      // RC522: RST -> D3
#define SDA_PIN D2      // RC522: SDA -> D2

MFRC522 rfid(SDA_PIN, RST_PIN);
byte ID[4] = {121,127,254,179};

void setup() {
  Serial.begin(9600);
  SPI.begin();
  rfid.PCD_Init();
  pinMode(RED,OUTPUT);
  pinMode(BLUE,OUTPUT);
  pinMode(GREEN,OUTPUT);
}
void loop() {

  if (!rfid.PICC_IsNewCardPresent())     // Yeni kartın okunması beklenir
    {
      BEKLE(); // Kart okutulmazsa mavi LED yanar
      Serial.println("okunamadi");
      delay(1000);
      return;
    }
 
  if (!rfid.PICC_ReadCardSerial())         //Kart okunmadığı zaman beklenilir
  {
      BEKLE(); // Kart okutulmazsa mavi LED yanar
      delay(1000);
      return;
  }
  if (rfid.uid.uidByte[0] == ID[0] &&     //Okunan kart ID'si ile ID değişkeni karşılaştırılır
    rfid.uid.uidByte[1] == ID[1] &&
    rfid.uid.uidByte[2] == ID[2] &&
    rfid.uid.uidByte[3] == ID[3]) {
    DOGRU();         // Okutulan kart doğruysa yeşil LED yanar
    delay(1000);
  } else {         // Okutulan kart yanlışsa bu blok çalışır
    YANLIS();         // Okutulan kart yanlışsa kırmızı LED yanar
    ID_OGREN();        // seri port ekranında kartın ID'si yazılır.
    delay(1000);
  }
  rfid.PICC_HaltA();
}
void BEKLE() {
      digitalWrite(BLUE,HIGH);
      digitalWrite(RED,LOW);
      digitalWrite(GREEN,LOW);
}
void DOGRU() {
      digitalWrite(BLUE,LOW);
      digitalWrite(RED,LOW);
      digitalWrite(GREEN,HIGH);
}
void YANLIS() {
      digitalWrite(BLUE,LOW);
      digitalWrite(RED,HIGH);
      digitalWrite(GREEN,LOW);
}
void ID_OGREN() {
  Serial.print("ID Numarasi: ");
  for (int sayac = 0; sayac < 4; sayac++) {
    Serial.print(rfid.uid.uidByte[sayac]);
    Serial.print(" ");
  }
  Serial.println("");
}
```

### Expected behavior:
13.56MHz RFID kartlar RC522 modülüne yaklaştırıldığında seri monitörde kart UID'sinin yazdırılması beklenmektedir.

### Screenshots:
[image]

### Additional context:
Boş bir projede yalnızca seri monitör üzerinde yazdırma yapma işleminde problem yaşamadığımı teyit ettim.
Kullanmakta olduğum RC522 modülü Arduino Uno'ya bağlayarak aynı programı başarıyla çalıştırabilmekteyim. Dolayısıyla modülde sorun yok.
Kullanmakta olduğum jumper kabloları test ettim, sorun yok. Ayrıca kart üzerindeki farklı 3.3V ve GND pinlerini de denedim, hala sorun devam etmekte.

zeynepdicle, 2022-10-31:
@MrDeadLine  Deneyap Arduino kütüphanesindeki PFIDkartOkuma (Examples(Örnekler) -> SPI -> PFIDkartOkuma) örnek uygulamasını deneyebilir misiniz ?

MrDeadLine, 2022-10-31:
> @MrDeadLine Deneyap Arduino kütüphanesindeki PFIDkartOkuma (Examples(Örnekler) -> SPI -> PFIDkartOkuma) örnek uygulamasını deneyebilir misiniz ?

Denedim ancak aynı problemi yaşamaktayım. Pin bağlantılarımda bir problem var mı acaba?

zeynepdicle, 2022-11-02:
@MrDeadLine Tekrardan merhaba,

Repodaki RFIDOkuma örneğini Deneyap Mini  Kartı ile gerçekleştirdim. Koddaki değişiklik ve yapmış olduğum pin bağlantılarını aşağıdadır. Bu değişiklikleri yapıp tekrar deneyebilir misiniz ?

**_MFRC522 mfrc522(D3,D2);   //D3:SDA(CS), D2:RST_**

RC522 -> Deneyap Mini
SDA -> D3
SCK -> MC 
MOSI -> MO
MISO -> MI 
GND -> GND
RST -> D2
3.3V -> 3.3V

MrDeadLine, 2022-11-05:
> @MrDeadLine Tekrardan merhaba,
>
> Repodaki RFIDOkuma örneğini Deneyap Mini Kartı ile gerçekleştirdim. Koddaki değişiklik ve yapmış olduğum pin bağlantılarını aşağıdadır. Bu değişiklikleri yapıp tekrar [...]

@zeynepdicle Merhaba,
Sıkıntının sebebi, SCK'nın Deneyap Mini'de SC'ye değil MC'ye bağlanması gerekmesiymiş. Desteğiniz için teşekkür ederim.

closed 2022-11-05 by MrDeadLine