deneyapkart/deneyapkart-arduino-core #16: Arduino IDE 2.0.* de kod derlenmiyor

emrullahjack, 2022-11-12:
### Donanım:
Kart | : | Deneyap Kart
---- | - | ------------
Kütüphane versiyonu | : | 1.3.8 
IDE | : | Arduino IDE
İşletim sistemi | : | Pardus 21.3
Upload mode | : | auto
Upload speed | : | 921600 
CPU frequecny | : | 240MHz
Flash frequency | : | 80MHz 
Flash mode | : | qio 
Partition scheme | : | default

### Hata tanımı:
Arduino IDE 2.0.1 ya da 2.0.2 (nightly build) ile deneyap kart için kod derleyemiyorum.

### Sorun yaşamış olduğunuz en temel kod bloğu:  
```cpp
// Kod farketmeksizin aynı hatayı veriyor, aşağıdaki koda özel bir durum değil.
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

**Hata ile karşılaşırken gerçekleştirdiğim adımlar:**
1. Yeni IDE yüklendi
2. IDE'yi ilk yüklediğimde sıkıntısız olarak deneyap kart'a kod derleyip yükleyebiliyordum
3. Bilgisayarımı kapatıp daha sonra tekrar açtım
4. Kod derlerken hata almaya başladım. Ayrıca örneklerde Deneyap Kart'a özel örnekler gözükmüyor.

### Beklenen çıktı:
Kodun normal bir şekilde derlenmesi. Zaten eskiz ile birlikte gelen varsayılan (default) kodu bile derlemiyor deneyap karta, hata veriyor kod farketmeksizin. 

### Hata mesajı:
```
No connection established

Compilation error: No connection established
```

### Ekran görüntüleri:
Seri terminal ekranında gözüken tek şey var o da yukarıdaki metin. Ekran görüntüsü gerektiren bir durum yok.

### Ek kaynak:
Yok

DogushC, 2022-11-14:
Merhabalar,

Arduino IDE'de menünden File (Dosya) --> Preferences (Tercihler) bölüme geldiğinde aşağıdaki bağlantı linkinin ekli olduğunu kontrol ediniz.
https://raw.githubusercontent.com/deneyapkart/deneyapkart-arduino-core/master/package_deneyapkart_index.json

Tüm detaylar için https://docs.deneyapkart.org/tr/content/contentDetail/deneyap-kart-arduino-ide-ile-programlama-dydk ziyaret edebilirsiniz.

[image]

emrullahjack, 2022-11-14:
Bu link ekli. Belirttiğim gibi bilgisayarı kapatıp açmadan öncesine kadar sıkıntısız yeni IDE yi kullanıyordum sonrasında bu hata ile karşılaştım. Ayrıca esp32 tabanlı kartların çoğunda "Request getBoardDetails failed with message: 14 UNAVAILABLE: No connection established" hatası veriyor. Aşağıdaki linkte belirtilen çözümleri uyguladım fakat sonuç değişmedi.

https://github.com/espressif/arduino-esp32/issues/7419

DogushC, 2022-11-14:
Arduino IDE 2.0.1' de sağ alt köşede hem kütüphaneler hem de kart altyapıları ile ilgili bildirimler geliyor; varsa burada belirtilenen güncellemeleri yapalım. Aktif bir şekilde Arduino IDE 2.0.1 kullanıyoruz; hem örneklere erişimde hem de kod yüklenmesinde bir problem görünmüyor. 
[image]
[image]

emrullahjack, 2022-11-14:
Anladığım kadarıyla linux'a özel bir durum var, sadece deneyap kart için değil diğer espressif tabanlı kartlar için de. Şuanda geçici çözüm olarak kart yöneticisini kaldırıp tekrar yüklüyorum bilgisayarım başlattığımda, ve normal bir şekilde çalışıyor.

**CLI'da gözüken hata:**
`daemon INFO panic: runtime error: invalid memory address or nil pointer dereference`

Araçlar > Kart > Deneyap (Ya da diğer esp tabanlı kartlar) seçildiğinde ilk hata veren yer burası sonrasında zaten IDE ile CLI arasındaki connection kırılıyor.

DogushC, 2022-11-14:
Geçici olarak da olsa çözülmesine sevindim. Bizde Linux'da deneyeceğiz; aynı sorunu yaşarsak https://github.com/espressif/arduino-esp32 ya da https://github.com/arduino/arduino-ide üzerinde konu (issue) açarız. Sizde bildirebilirsiniz.
Katkınız için teşekkürler.

emrullahjack, 2022-11-17:
Hızlı geri dönüşünüz için ben teşekkür ederim. Evet ben de fırsat bulunca https://github.com/arduino/arduino-ide üzerinde konu açmayı planlıyordum. IDE'yi yazan arkadaşların C/C++ kodu ile alakalı bir sıkıntı var gibi duruyor. Ya da kullandıkları API'lardan birinde uyuşmazlık oluşuyor. 

En azından şuana kadar problemin kaynağı hakkındaki olasılıklar kümesini daraltmış olduk. Teşekkür ederim.

DogushC, 2022-11-17:
Katkılarınız için teşekkür ederiz.

closed 2022-11-17 by DogushC

emrullahjack, 2022-11-17:
Son olarak bir bilgilendirme yapayım, hata Arduino IDE kaynaklıydı ve giderildi, gün içerisinde 2.0.2 versiyonu yayınlandı. Sıkıntısız çalışmakta.