deneyapkart/deneyapkart-arduino-core #19: Arduino Upload failed error 2

faizannazir, 2024-03-07:
ide arduino 2 upload failed error 2 without flash it also printing error on serial monitor when flash failed it stop printing that error 

on flash verbus thuis print tried with different upload rate same error
 Serial port COM4
Connecting...
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:aa:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00047fff...
Compressed 13200 bytes to 9522...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)
Failed uploading: uploading error: exit status 2

pxsty0, 2024-05-22:
Hello, if you can send us your support record via the forum (https://forum.deneyapkart.org/) or destek@deneyapkart.org e-mail address, we will help you.

closed 2024-05-22 by pxsty0

faizannazir, 2024-05-22:
The issue was device does not have flash memory so i replaced the esp32 c3 with flash memory one and it works fine